parallaxinc/Flight-Controller #6: Filename cases are wrong for Firmware-C

electrodude, 2016-04-06:
Most operating systems other than Windows (e.g. OSX and Linux.) have case sensitive filesystems.  The C firmware does not compile on these systems because the cases of the filenames on disk and the filenames in `#include` directives do not agree.

Fixing this would be as simple as either renaming the files or changing the `#include`s so that they agree with each other.

electrodude, 2016-04-06:
I would submit a pull request to do this for you, but I'm not sure whether you'd rather rename the files or change the `#include`s.

JasonDorie, 2016-04-06:
I'll take care of it - thanks for letting me know.  I think I've done that
for all the GroundStation files already, but I've never tried to build the
firmware on anything but a PC.

J

PropGit, 2016-04-06:
Thanks for taking care of it, @JasonDorie.

JasonDorie, 2016-04-07:
Electrodude - If you can easily test this, sync and let me know if I missed
anything.  I believe this should be correct now.

electrodude, 2016-04-15:
It compiles, but I didn't actually test it yet.  I'm closing it now.  Thanks.

closed 2016-04-15 by electrodude