parallaxinc/Flight-Controller #11: "Remote RX" Receiver Type not functional

RoboTechie, 2016-07-25:
As tested with a Spektrum AR7700 receiver, the _Remote RX_ receiver type, listed on the _radio setup_ tab of Parallax Groundstation, does not work. There is no response to transmitter stick/switch movements; all radio channel values remain constant, as shown in the screen capture below. I recommend the Remote RX receiver option be removed/disabled from GroundStation until it is proven to be functional.

[image]

JasonDorie, 2016-07-25:
I flew it numerous times with an actual remote receiver, so I know it
works.  It's probable that the 7700 has a different ID value or something
in the header and is being filtered out as a result. I can look at it once
I have my computer set up again.  We moved into our new non-temporary place
last night, and things are chaotic, but we're making decent progress on the
unpack.

J

MatzElectronics, 2016-09-12:
A couple of things related to this - I started playing around with a satellite and getting data, and I noticed a couple of things: 1) none of the ID/packet headers online matched what I ended up having to use (I ended up using 0x5a00 with the MSbit being masked off), so there may be something to the ID varying between devices. 2) Binding a satellite to the TX is challenging without having a RX to plug it into when binding - something to explore doing with the GS, but that's a whole 'nother issue. 3) when the radio turns off, the packets stop coming.  No failsafe...nada.  That means we need to code one in for that receiver type...

JasonDorie, 2016-09-12:
I flew my little racer with just the remote and the range was very
short, when it got out of range it never re-connected.  I wouldn't feel
comfortable with someone flying one of those unless it was very short
range.  Might be worth looking into if there's a protocol to follow after
signal loss or something - I was surprised that I had to power cycle it to
wake it back up.

Binding is another issue - the bind protocol is documented somewhere, but
it's very timing sensitive - something like within 200ms of power on you
have to send a specific pulse sequence to the RX, so the only way to
support it would be either hardware changes, or to have a power-on bind
mode for the FC itself.  If the Propeller boot-up takes longer than 200ms
(or whatever the number is - I don't remember) then that method won't be
possible.

J

MatzElectronics, 2016-10-20:
Fix in issue #48.

closed 2016-10-20 by MatzElectronics