parallaxinc/Flight-Controller #12: GroundStation display issues with Windows 10

RoboTechie, 2016-07-25:
I have found that some text overlap/clipping can occur when using Parallax GroundStation v1.1.0 on Windows 10, depending on display settings.

Appearance of "Radio Setup" tab on a 1920 x 1080 display with default display settings (no text/app resizing):
[image]

Appearance of "Radio Setup" tab on a 1920 x 1080 display with "Change the size of text, apps, and other items" (settings>system>display) set to 125% default display settings (no text/app resizing):
[image]

As the magnification is increased further, the issue worsens and effects more tabs. At present moment, it appears as though that windows setting only effects text size in GroundStation, all other graphics/boxes remain the same size.

JasonDorie, 2016-07-25:
This is exactly the problem I'm in the process of trying to fix.  It's the
same problem as the text issues on Mac and Surface devices - text scales
based on user prefs but the window & layout doesn't adjust accordingly.
I'm about 1/2 done with it.

J

RoboTechie, 2016-07-25:
That's what I was hoping was the case. I just upgraded my machine to Windows 10, and this was the first time I had personally experienced the issue, so I figured I'd share the screen caps to double-check that it was the same issue, and to try out the issue tracker feature.

VonSzarvas, 2016-07-25:
To add some feedback about this on Windows 10. I've had similar issues with cad, paint and pcb software after going from Win2012 to Win10.

The windows scaling system does seem to break some apps, and more so as resolution increases. 
I certainly can't use 4K screens reliably on Win10, which is a bugger when working with large schematics and drawings :)
3840x2160 worked great until win10 came along... poop!!

It's not just text, but text boxes and fixed size elements (panels, etc..) don't handle well at 4K either ! Often remaining at half the size required to view the content!

Kyle- temporary fixes might be:
1) Adjust the windows text size from whatever it's set at (150%?) down to 100% (although that's not really pratical when you get to resolutions above ~ 2000x1500 :)
or
2) Right click the executable of the program your interested in, go to compatibility settings, and disable scaling for that app.

Have fun solving this one Jason ! 
I'm gonna guess this is one of those tasks that SHOULD be trivial, but ...... :)

M.

JasonDorie, 2016-08-07:
This should now be fixed.  All panels now handle auto-layout properly, and I shortened the channel labels to reduce the likelihood of overlap in the first place.  I've verified on my own desktop machine, but please check on others, and an MS Surface if you have one.

VonSzarvas, 2016-08-08:
Win10
Resolution: 3840x2160

Win10 scaling set at 125% looks ok.
Win10 scaling set at 150% (which is the recommended default according to the Win10 install I'm using) looks broken.

Specifically, at 150%, there are 2 elements with issues:

1) The menu tabs
2) The "group-box" titles

Group-box is VS terminology- no QS here :)
The issue as it appears to me with those, is that the text gets larger, but the width allocated for the title text seems to remain the same, and oddly the title text appears to center inside the available width (so the text is cutoff at both ends). They looked more like align-left at 125% ?!!?!

Attached last screengrab of an enlarged GS window showing the issue nicely. Oh, and those values to the right of the sliders seem to be pushed out of space too.

Overall- I'm not entirely certain this should be a priority at all..... Probably not many schools running on 4K screens yet. Maybe this feedback is something useful for the future though. And in that time, perhaps MS will also improve scaling handling.

M.

RoboTechie, 2016-08-08:
The latest release (v1.1.0 gamma) is a significant improvement, on my 1920x1080 external monitor (windows 10), there are no issues at 100% (1.0x) or 125% (1.25x) scaling! (which there were issues with previously). However, there are still issues at 150% and 175% scaling, just as Michael reported. I've attached photos of the radio tab at 125%, 150%, and 175% to show the issue. But, as Michael said, with the issue fixed for 100% and 125% scaling, fixing it for the higher magnifications is a lower priority.

[image]
[image]
[image]

RoboTechie, 2016-09-19:
Why don't we replace the Parallax logo in GS with a Vector format (or very high res) so that it also scales with the window?

JasonDorie, 2016-09-20:
Because it didn't seem like the most pressing need at this point.  :-)

RoboTechie, 2016-09-20:
haha I don't disagree. But in terms of feasibility/difficulty, where does this stack up, and what would be the better option, a vector format file (so I assume it would have to be just line art?), or very high res file?

JasonDorie, 2016-09-20:
It's probably not difficult, but it's a) not broken and b) an hour I could
spend working on the compass or other more impactful features.

UI changes are fussy - you have play with things and iterate until it
works.  Changing the scaling of that single element is probably not hard,
but I don't know how QT handles images and scaling so I'd have to play with
it to find the right settings, or write it custom.  To take an image of a
given size and display it was trivial.  Getting it to scale and auto-layout
could take 10 minutes or an hour, depending on how much I have to mess with
it.

That image also contains texture in the background, so it's be better
suited to a bitmap.

J

JasonDorie, 2016-10-05:
I investigated this.  Qt has no built-in facility to maintain the aspect of an image when scaling it, so it requires writing a custom control.  It won't be hard, but it's kind of a waste of time.

MatzElectronics, 2016-10-05:
Not worth it at this point.

closed 2016-10-05 by MatzElectronics