parallaxinc/Flight-Controller #19: Add hexacopter type to groundstation

MatzElectronics, 2016-09-17:
This is paired with issue #18.  Groundstation needs to adapt to hex configuration.  It may be possible for the GS to auto-detect this, either through a modified heartbeat, or by detecting that the motor speeds packet is longer.

JasonDorie, 2016-09-18:
I have done some of the work toward this:

- GroundStation now has two more motor buttons (they are hidden currently
at startup)
- GroundStation and the FC now use different commands for motor, LED, &
buzzer test, and throttle calibrate (this was necessary to allow for
nudging two more motors, and is just cleaner is general)
- The two additional motor buttons DO send the correct motor nudge
commands, but they'll be ignored by the FC unless you have 6 motors
connected
- I've added defines in the header to set the flight mode to QUAD or HEX
- I'm currently updating GroundStation to detect the motor count and adjust
accordingly

MatzElectronics, 2016-09-18:
@JasonDorie I'm flying a hex currently - I have it in a feature branch of
my local repo - I'll move it into a feature branch of the main repo, and we
can figure out how to merge what's all there.
Thanks!

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
> auth/ANak_iRbb9lpxoZrSbm_

Don't put it in main - I'll be checking this all in to the ActiveDev branch
in a few minutes.  Then you can compare mine with yours and fix whatever I
broke.  :)

J

MatzElectronics, 2016-09-18:
Sounds good!

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
> auth/AS0quG8H13zHzFCrg-

FYI - My motor outputs are:

AUX1 - Hex-right
AUX2 - Hex-left

I'm assuming a config that's basically an X with an extra motor in the
middle of each side, and those motors are HR and HL.

J

MatzElectronics, 2016-09-18:
For some reason I can't brain that :)  If you aren't too far in - Motor 1
is front left/CW (same as Quad), and then they increase CW around the hex,
alternating in spin.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
> auth/ANak_hqrljzid4AK_

Too late - submitted.  Pull the ActiveDevelopment branch, set the
FLIGHT_MODE_HEX define in the header, flash your quad, build
groundstation.  You'll see the numbers for the motor outputs in the code,
and all the buttons work.

J

MatzElectronics, 2016-09-18:
Ok - Kyle and I may change the motor order - but I'll take it for a drive
hopefully tomorrow but for sure in the next day or two.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
> issues/18

The reason I didn't do them sequentially is because it means more buttons
in UI that have to be reconfigured / relabeled when it changes.

The way I have it, motors 1 to 4 are always the same index / port.  Front
left / front right always spin the same way, and only the other change.

On a hex, the front left / back left spin the same direction, on a quad
they don't, so I have to rearrange the icons on the buttons when the UI
changes modes.

Does that make more sense?

MatzElectronics, 2016-09-18:
> auth/AS0quGyEr1LGfMlSIImZ-

Yes - sorry, long day!  I'm game for leaving it - but I'm mildly concerned
about the rotation direction of pairs of motors being the same that way.
Not sure that it matters, but, still, my inner neat freak is cringing
slightly :)

Also, I did see that you have Roll? at 3/4 - divide by 2 works really well,
and I think I did the trig on that a while back...

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
Uh, yeah, I was thinking 45 degrees, not 30 - the divide by two is very
probably correct.

J

MatzElectronics, 2016-09-18:
Yeah, my napkin doodle says so.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>