parallaxinc/Flight-Controller #21: In-depth PID research and tuning

MatzElectronics, 2016-09-17:
Research/experiment with (and most importantly document the effects of) the full range of all individual PID settings.  From there, optimize for the standard configuration, and make recommendations of common changes/configurations.

MatzElectronics, 2016-10-20:
From @RoboTechie / #45 : 
While disabling the median filter on the SF11-C (Instructions here) fixes the catastrophic oscillations described in Issue 13, the ELEV-8 still oscillates 10-50 cm vertically without ever settling out when using the SF11-C for altitude hold, so any improvements to reduce this would be well received.

JasonDorie, 2016-10-20:
Is this with a big battery or a gimbal, or the 3300 battery and no
additional hardware?  (I agree it needs tuning, but I seemed to have fewer
issues than reported).

MatzElectronics, 2016-10-20:
Ideally a comprehensive and deep dive into this - multiple configurations, so we know how various settings, loads, and configurations impact different aspects of the craft's performance, both in terms of setpoints and range.

JasonDorie, 2016-10-20:
I meant specifically what config are you flying where you're seeing the
oscillation?

RoboTechie, 2016-10-25:
I was using a 5300mah battery (which I consider the "standard" for the v3)
other than that, the only other payload was the laser sensor and a ping
sensor.

> Ideally a comprehensive and deep dive into this - multiple configurations,
> so we know how various settings, loads, and configurations impact
different
> aspects of the craft's performance, both in terms of setpoints and range.

> <https://github.com/parallaxinc/Flight-Controller/issues/21#issuecomment-
254999674>,
tBDMh7dqC_GUhE6ks5q1t8TgaJpZM4J_gs4>

JasonDorie, 2016-10-25:
> issues/21#issuecomment-
> 255000267>,

Ok, good to know.  I typically fly with the 3300mah.