parallaxinc/Flight-Controller #24: Firmware expo on channels 2, 3, 4 + throttle change limiting?

MatzElectronics, 2016-09-17:
This is a "tuning for beginners" concern.  Most v3's are sold with DXe's, which require a separate cable, software, and are generally a PITA to alter their settings.  This is a discussion about whether expo + capping/softening sudden throttle increases is needed.

JasonDorie, 2016-09-17:
I have often thought expo would be nice to do in the controller itself, and
it could be done reasonably easily with a lookup table.  Would cost some
time & memory, but maybe not too bad.

Throttle change limiting also would be relatively easy - could make it so
you can't change it more than X per update.  Would obviously be a beginner
setting, but also worth trying.

J

MatzElectronics, 2016-09-17:
For throttle change limiting, I'd prefer limiting rapid increases only...and the amount of limiting as a settable option.
For expos - would doing it by formula take too many cycles?  I'm sure we could find a way that's a good balance of memory vs. cycles.  I played around with some options on a spreadsheet, and this looked good:

Radio.Aile = Radio.Aile / ( 1024 / ( abs( Radio.Aile ) ) )

Just have to watch out for divide by zeros.

Another option would be two or three linear approximations with increasing slope...

JasonDorie, 2016-09-17:
Your formula is fine, but it won't work without using float math, and
that'll make it really slow.  I was thinking a 16 or 32 entry table with
linear interpolation in between, or possibly a quadratic curve function,
evaluated in fixed point.  Probably slower to do the latter, but I'd test
both and see.

For throttle limiting what I was thinking was just limiting the amount of
change from one frame to the next, but I can see value in allowing rapid
cuts - I'll think about it a bit.  Shouldn't be too hard.

J

MatzElectronics, 2016-09-17:
I'm good with the small table and interpolation.  I'll draft it and test it out.  I can take a crack that the throttle up limiting too.

MatzElectronics, 2016-09-18:
I tried flying with expos for the first time today - and that makes a WORLD of difference.  40% felt about right - so yes, we definately need to code this in.

JasonDorie, 2016-09-18:
Yeah, there's a reason this is a standard radio feature.  :)

MatzElectronics, 2016-09-18:
Except setting on the DXe is a PITA.

JasonDorie, 2016-09-18:
> auth/ANak_uUYsiZ3R2iwKnYv-

Expo on Roll/Pitch and Yaw (two settings) is submitted.

MatzElectronics, 2016-09-18:
Moving throttle change limiting to a separate issue

JasonDorie, 2016-11-13:
Tested, working

closed 2016-11-13 by JasonDorie