parallaxinc/Flight-Controller #31: File output of recorded sensor data

MatzElectronics, 2016-09-17:
Preferably as a .csv format file with the first line being the column headers.

JasonDorie, 2016-10-05:
Completed, in ActiveDevelopment branch

jrleeman, 2016-10-16:
Is there an automated build of the ActiveDev branch? I'd like to use/test this functionality.

JasonDorie, 2016-10-16:
There isn't, no.  I disabled the auto-build we had because it kept failing.

jrleeman, 2016-10-19:
I did play with this before I borked my setup. Any way to make this a checkbox instead of a menu option. i.e. you could "enable logging" and stream to a file during the entire flight?

JasonDorie, 2016-10-19:
Kind of, but I don't actually know in GroundStation when it's "flying".  I
could stream to disk when the motors are spinning, but if you disarm /
rearm and fly again, do I append, overwrite, or increment to a new file?
What would your preference be?

J

jrleeman, 2016-10-19:
I'd be okay if it logged the entire time the box is checked - flying or not. Maybe that's just me.

What about naming the files with a timestamp like DDMMYY-HHMMSS.csv or something similar?

MatzElectronics, 2016-10-19:
I like the idea of a datestamp filename if it's not too much trouble - I
would recommend YYYY-MM-DD_HH-MM-SS.csv.  Longer, but then when you sort
the files alphabetically, it also sorts them in date order.

jrleeman, 2016-10-20:
That could work. What about adopting the naming convention used on seismograms and other geophysical data? YYYY-MM-DDTHH:MM:SS (i.e. 2016-09-03T12:02:22)

MatzElectronics, 2016-11-15:
Functionality is tested and verified.  Will leave open for two pieces: Some reference to time (cycle, phase, anything that time can be inferred from), and default file naming if possible.  If it's not possible or easy to do, we can skip that part because the user has the ability to name the file themselves.