parallaxinc/Flight-Controller #32: Serial command set

MatzElectronics, 2016-09-17:
Serial Command Set or other method of control outside of transmitter.  I'm guessing that this is a precursor or building block to GPS control anyway?  It is the most requested feature after GPS itself.

JasonDorie, 2016-09-18:
I would not recommend this until GPS is implemented.  The quad has no way
of holding a set position otherwise.

I mean, I could do "set roll", "set pitch" style commands, but those have
no way to know if you're actually moving forward, sideways, etc.  With GPS
position hold, we'd just change the hold position (forward 10cm, back 10cm,
etc) and the rest of the system would just deal with it.

J

MatzElectronics, 2016-09-18:
I do think it should include both - partially because of GPS denied
environments, and partially because this is forward thinking with using a
safer mini-quad.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-09-18:
So my next question is, how do you see users feeding these commands in?
Like, from a terminal?  From a homebrew XBee radio?  From within the
firmware itself?  Just curious what you're trying to accomplish by
providing these commands.  I really think without any sort of position hold
it requires a pilot, so I'm wondering what you foresee this being used for.

MatzElectronics, 2016-09-18:
The RX/TX port, with another micro issuing them.  It's a request we get
from universities most frequently who are doing things with computer vision
or some other kind of research, and are using the v3 as a base platform
because it's a good workhorse for that type of thing.