parallaxinc/Flight-Controller #36: Slow flight loop?

MatzElectronics, 2016-09-17:
The current flight loop is 4ms - should it be increased to 5ms to accommodate the additional math required to GPS, and a few additional features?

JasonDorie, 2016-09-18:
Done.  Not submitted, not flight tested, but IMU appears to be working
properly.

I've also reduced a bunch of the data being sent to GroundStation,
including the radio channels, the yaw / pitch / roll power values, counter,
and one or two others.  Since the update rate is now 20% slower, it means
we have 20% more time to send serial data.  With the reduction in packet
size, we should have more overhead for serial retries, and generally lower
packet loss.  It also doesn't hit the FTDI buffer thing as much either.

I figure since this version won't be back-compatible anyway, I might as
well break *everything* at once.  I bumped this to version 3.0.0.

J

MatzElectronics, 2016-09-18:
Thanks - I figured that would be the case :)

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

JasonDorie, 2016-11-13:
I've tested this on the small quad and it's working, so the big quad will be fine.

closed 2016-11-13 by JasonDorie