parallaxinc/Flight-Controller #38: Bidirectional data link between FC and GS

MatzElectronics, 2016-09-17:
Give the user a path to send data from the FC to the groundstation - maybe a typedef that looks like user.fc_to_gs1, user.fc_to_gs2, possibly as bytes, with no expectations on refresh rates.

Also, in the other direction - GS to FC, in the same manner, user.gs_to_fc1, user.gs_to_fc2

This is a request to build the pipes, and possibly a tab on the groundstation, and/or incorporate them into the sensor graph

JasonDorie, 2016-09-18:
For the "to the FC" version, what is the value?  Int? Float?  What range?
How does the user enter it?

I can add an Int or two from the FC to the GS and display them, but the
other way is kind of specialized.

J

MatzElectronics, 2016-09-18:
It can be two chars (ints).  I actually did it by creating a COMMAND, but
only checking chars 1 and 3, and repeating the value on chars 2 and 4.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>