parallaxinc/Flight-Controller #40: Telemetry Lockups

jrleeman, 2016-09-17:
Using the 2.0 firmware and 2.0 ground station on Windows 10 the telemetry will lock up every few minutes for up to a minute before restoring. This has been a problem in the last couple of releases.

MatzElectronics, 2016-09-17:
@jrleeman This is probably related to issue #23.  Can you describe your setup?  Are you connecting through XBee or USB?

Currently, there is no handshaking, so the idea is to keep the packets small and coming in a steady enough stream that they don't overflow the buffer on the FTDI chip.  Windows also has a bad habit of de-prioritizing the FTDI driver, which means data isn't getting pulled frequently enough to prevent buffer overrun.  It helps to close any other programs that are running, and see if that makes a difference

I'll mark this duplicate for now.  Thanks for the input!

jrleeman, 2016-09-18:
The lockup happens though XBee only in my testing. This is running on a low-end surface with Windows 10. I'd be a bit more hesitant to mark this as a duplicate. There were no other programs open and I can't find a reliable time interval between lockups. Also as @JasonDorie pointed out I'm unsure that handshaking would do anything here.

JasonDorie, 2016-09-18:
I don't have a Surface, or anything running Windows 10, so I have no means
of testing or debugging this.  Does the bottom of the screen still say
"Connected" when the telemetry isn't showing?

jrleeman, 2016-09-18:
Yes - still says connected. I was just able to reproduce this error on a MacBook Air running 10.11.6. Telemetry runs for ~15 seconds, then locks up. If I power the ELEV8 down, the ground station still says connected. XBees are still talking from the comm activity LEDs.

MatzElectronics, 2016-09-18:
@jrleeman Are you willing to do some testing of an improvement in the dev branch?

jrleeman, 2016-09-18:
Sure - I don't have the GCS build environment setup. Is it built using QT Creator?

JasonDorie, 2016-09-18:
Yes, GroundStation is built with Mingw, Qt 5.5 (though newer should be
fine).

J

MatzElectronics, 2016-09-18:
Yes, it is

jrleeman, 2016-09-19:
Can you all duplicate?

MatzElectronics, 2016-09-19:
I won't be able to until I get back tomorrow

MatzElectronics, 2016-09-19:
@jrleeman It might take me a few days before I can set this up, but I'm going to clip on a serial monitor and watch all of the data going through - I'm starting to suspect that the xbee is going into command mode or getting "illegal" combinations of chars.  I'll keep everyone posted on what I find.

jrleeman, 2016-09-23:
@MatzElectronics any luck on just reproducing the behavior? From the activity LEDs I think data is still being sent, but haven't verified with a serial dump.

MatzElectronics, 2016-09-23:
Not yet.  I haven't been able to get this all set up yet as we have a
deadline on another project approaching.  My plan was to use a prop plug or
FTDI device to probe and capture the communication, and see if any clues
emerge from that.

jrleeman, 2016-10-07:
Any luck? I'm hoping to write this up for a Servo Magazine article, but want to make sure it is a working system first.

RoboTechie, 2016-10-12:
@jrleeman 
Sorry we've been so slow to respond; it's been very very busy around here. I can't imagine how frustrating it is that your telemetry is not working. I think one of the challenges is that we (or at least I) have not been able to reproduce. I set up telemetry on my ELEV-8 running firmware v2.0.0 connected to GroundStation v2.0.0 and left it running for a little over 7 minutes, and the telemetry never dropped (the sensor graph in this GS version does loop around at x=6000, but all of the values continue to update). Here are a few more thoughts:

1. If have the means to do so and haven't already, try a different computer, especially if the only one you have tried is a tablet.
2. Another possible cause is interference in or around  the 902 to 928 mHz ([33cm](https://en.wikipedia.org/wiki/33-centimeter_band)) band, as that's what this telemetry equipment uses. If there is a high level of noise, the data or heartbeat signal could get jumbled beyond recognition, casing the data to drop out. As far as evaluating interference goes, the easist step is to consider if you know of any possible signal source near you. Around this band, that may likely to include amateur radio activity, and radiolocation (as use by cell towers, among other things). I also recommend you try out the Spectrum Analyzer function of [XCTU ](https://www.digi.com/products/xbee-rf-solutions/xctu-software/xctu) to look for interference. You'll need to connect one of your XBees to the computer and connect to it in XCTU, then click on the wrench/screwdriver icon near the top of the XCTU window and select "Spectrum Analyzer" from the drop-down menu. In the SA tool, select the XBee connected to you computer, and then run the default analysis, which should take a few minutes. Here's what the results of my analysis look like:
[image]

In the meantime, I'll work on putting together and testing an entire Flight Controller + Telemetry system to send to you tomorrow if we don't get anywhere from trying another computer and checking for interference. 
Thanks for you patience!

JasonDorie, 2016-10-12:
>    minutes. Here's what the results of my analysis look like: [image:
>    image]

Did someone on this thread mention that it was repro'd on a non-windows
system too?  I seem to remember that, but can't find it.  I don't have a
surface or win10 system, and I've been trying to report but can't.  I've
left this running for 15 minutes (over XBee) and not seen any loss of
communication, and I've been through the code a few times now looking for
suspects, and can't see any reason it'd be dropping for that long.  A few
packets here and there I totally expect, but it should re-sync right away.

jrleeman, 2016-10-13:
I did reproduce on a mac as well. I’m on the road again, but will take a screen video this weekend and check for interference. Jim was having this same problem though.

JasonDorie, 2016-10-13:
I have a Mini Mac, and will have a Macbook next week, so I can try to repro
this on those.  It's very hard to diagnose if I can't reproduce the issue,
but I have a few ideas for things to try.  At the very least I can add a
logging file with some diagnostics and have you run that version and send
me the results.  It's not ideal, but it might be our best option if I can't
get it to happen locally.

RoboTechie, 2016-10-13:
Good news/Bad news. I borrowed Ken's xBee setup, and it's having the intermittent connectivity issue, so I should finally be able to start troubleshooting first thing tomorrow. But this indicates the issue is probably not related to your operating system or interference.

jrleeman, 2016-10-14:
I suppose that's good news! Looking forward to hearing what you find.

RoboTechie, 2016-10-14:
Ok, I've been able to re-create (and then fix) the issue by changing the _**CE** Routing/Messaging_ mode setting of base-station xBee using XCTU.

To see if that's what's causing your problem, try this fix:

1. Plug you base-station xBee into your computer 
2. Launch XCTU (the xBee configuration program) on your computer.
3. Click on the "Add Devices' Icon
4. Select the COM port corresponding to your xBee, and click "Finish". XCTU should automatically find and add the xBee to your radio modules list. (If it doesn't, give me a call at 916 625 3003 and I'll walk you through recovering the module)
5. Click on the module listing below _Radio Modules_ .  Wait for XCTU to load all of the module's settings.
6. First, scroll down until you find the setting for  _**BD** Baud Rate_, and double-check that it's set to 57600
7. Next, scroll back up to _**CE** Routing/Messaging_  and select _Non-Routing Module_ from the drop-down menu
8. Click the pencil icon to the right of the _Routing/Messaging_ setting or the Write icon in the top bar to save your settings.
9. Once XCTU confirms it has sucessfully made the settings change, close out of the program, and try connecting to your ELEV-8 again. (note that you should only be changing the base-station xBee to a _Non-Routing Module_, leave the xBee on your ELEV-8 as a _Standard Router_"

jrleeman, 2016-10-14:
@RoboTechie I'll give this a shot tonight/tomorrow morning and let you know.

jrleeman, 2016-10-15:
Ok - that does the trick it looks like. On my tablet (Windows 10) there would be very occasional and brief losses of telemetry, but that's possibly due to it's weak processor. Was maybe a few seconds every 5 minutes at most. MacBook Air had no issues at all.

JasonDorie, 2016-10-15:
I've done some reading and experimenting with my own XBee pair here and
come up with some additional recommendations:

- Use point-to-point mode instead of broadcast mode by setting the DH / DL
entry of each device to match the SH / SL entry of the other.  This makes
them talk only to each other and reduces comm overhead

- On the base station, leave the RR (retry count) setting as 'A' (10), but
on the flight controller, lower the retry count to 3.  We don't really care
about individual packets as much as more current data.

These two settings reduce the latency from FC to PC significantly.

You can also bump the baud rate to 115,200 on both sides.  This means
changing the serial init of the FC firmware to initialize the 2nd port as
115200 instead of 57600.  It does not change the actual throughput, because
that's limited by how often I send the packets, but it means that the XBee
itself gets the data a little quicker, and is more likely to packetize
properly because there's more time between packets.  I'm likely to make
this the new default setting - I initially thought this was the comm rate
between XBees, but it's actually just the comm rate between the host and
the XBee.  The XBees communicate at their own rate regardless, and the
throughput is the important bit.

MatzElectronics, 2016-10-15:
Interesting.  I'm anxious to try this out over the weekend!

hurricanehover, 2016-10-15:
Jason. Just reprogramed XBEE's. Looks like it works well. 30 minutes and no dropouts.

jrleeman, 2016-10-17:
I had it working well - after making all changes (and playing around compiling v3.0 of GCS) I have a mysterious issue. I've gone back to V2.0.1 everything and upon connecting to the quad the indicators go away (see screenshot). It still responds to remote commands, but most of the telemetry isn't showing. 
[image]

MatzElectronics, 2016-10-17:
> [Uploading Screenshot (17).png…]()

Good evening,

Did you reinstall version 2.0.1 of the FC firmware back on your v3?  The
two FC firmware versions are not compatible...

jrleeman, 2016-10-17:
Yes (visible in the screenshot, both are running 2.0.1)

MatzElectronics, 2016-10-17:
Sorry!  I can't see the screenshot from my phone.  Jason may have to answer
this one...

JasonDorie, 2016-10-17:
Hmm - I haven't seen that before.  I'll take a look tonight.  I flip
between the ActiveDev and another branch quite a bit, and was back on 2.0.1
to release it.  Do you have a saved prefs file you can re-apply?

jrleeman, 2016-10-17:
No, I don’t. 

I may blast everything clean and try again if you can’t reproduce.

JasonDorie, 2016-10-17:
Try bumping back to the other version, use GroundStation to save a prefs
file, revert back to 2.0.1 and apply it.  (You'll need to go to one of the
screens with an "upload" button to apply the loaded prefs).

That might not work, but the prefs are written is such a way that they are
mostly compatible between firmware versions.  You should also just save the
prefs from your 2.0.1 firmware and attch the file - I'll take a look and
see if anything looks wrong.

jrleeman, 2016-10-19:
I had already deleted V3. Bad news is that the problem is replicated on my Mac which has never had anything but 2.0.1. I re-downloaded a clean set of firmware and set the XBees back to 57600. No luck. Should we close this issue since telemetry is indeed flowing, but open a new issue for the display?

JasonDorie, 2016-10-19:
It'll be an issue with the format of the data coming from the Elev8, most
likely.  Everything is responding incorrectly?  IE, radio movement,
orientation display, sensor graph, or just some of those?

When you re-flashed the firmware, you programmed to EEPROM, not just RAM,
correct?

J

jrleeman, 2016-10-19:
Yes - EEPROM programmed. I noticed my radio settings carried over. Settings file attached.

Sticks on screen and some telemetry are fine, others like battery are not working.

[JRL.zip](https://github.com/parallaxinc/Flight-Controller/files/537739/JRL.zip)

JasonDorie, 2016-10-19:
I'll take a look after dinner and see if I can spot anything.  Meanwhile,
make sure you're loading / flashing the right firmware and running
GroundStation from that same location.  It's possible if you grabbed an
active-dev version that the version number hadn't been update but there
were format changes.  Sometimes the version number change happens after
other stuff.

J

JasonDorie, 2016-10-19:
More to the point, what you're suggesting *shouldnt* be possible with
release versions, so just double check in case it's something simple and
the UI is mis-reporting the version no.

jrleeman, 2016-10-19:
Okay - I reset all factory defaults and reflashed. Indicators are back working now. Sorry for all the back and forth. Hopefully the data format is settling down soon? 

w.r.t. trying future dev stuff. Is master working generally and all development in ActiveDev?

JasonDorie, 2016-10-19:
Going forward, yes.  Master should always be working and stable.  ActiveDev
will be in-flux, brought to stability, then branched across to master.

J

jrleeman, 2016-10-19:
Happy to contribute to the write-up/tutorial if there's an easy way to. Is that content on GitHub somewhere or does it live on a Parallax server somewhere?

JasonDorie, 2016-10-19:
Probably not - that would all exist within Parallax.  Matt's baby.

J

MatzElectronics, 2016-10-20:
Going to hold this open until I edit related docs, once there, I will close.

JasonDorie, 2016-11-13:
Assigned to Matt as he's holding open until docs are updated.

jrleeman, 2017-03-08:
Did the parallax docs ever get updated on this? Servo released the article as a free and open to all article: http://www.servomagazine.com/index.php/magazine/article/January2017_MultiRotorHobbyist_Telemetry

hurricanehover, 2017-03-08:
Hey John. Looks like it's still available:

 

https://forums.parallax.com/discussion/165345/esp8266-drone-atmosphere-sound
ing

 

 

  _____  

 

http://www.servomagazine.com/index.php/magazine/article/January2017_MultiRot
orHobbyist_Telemetry

-
<https://github.com/parallaxinc/Flight-Controller/issues/40#issuecomment-285
096437>  it on GitHub, or mute
Aey3zpGks5rjtuZgaJpZM4J_r6G>  the thread.
<https://github.com/notifications/beacon/AH_3BAQK6MXXrikSfFBlKH2bWFZCUojCks5
rjtuZgaJpZM4J_r6G.gif>

jrleeman, 2017-03-08:
Right - I mean the online "assembly guide/tutorial" *from Parallax* that needed some revision after the issues we found troubleshooting this.

MatzElectronics, 2017-03-09:
Good morning,

No, unfortunately I (and most of our team responsibile for documentation)
have been pressed to meet a deadline for the BlocklyProp project, and it
may still be a few weeks before I am able to circle back to some
documentations needs for the ELEV-8.  This need, as well as a couple others
will be addressed, but we haven't been able to get there recently.
-M

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

jrleeman, 2017-03-10:
No worries - I was just going through cleaning up issues, etc I'm involved with :) 

@hurricanehover It was mostly the setup of the XBee radios that we changed I believe.