parallaxinc/Flight-Controller #41: Mac Ground Station Versioning and Telemetry

jrleeman, 2016-09-17:
In the 2.0 release the ground station is still labeled "1.1.1" in the footer of the program and will not connect to a telemetry system that works well on the Windows version of the ground station.

MatzElectronics, 2016-09-17:
@jrleeman have you tried reinstalling the FTDI driver for the mac?  I think the next rebuild of 2.0.0 corrects the version on the bottom - @JasonDorie can you verifiy that?

jrleeman, 2016-09-18:
New FTDI VCP solved this. Not sure why it was necessary. Now just to get that version number.

MatzElectronics, 2016-09-18:
Yes, that's coming - thanks!  I'll close this issue.

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>

closed 2016-09-18 by MatzElectronics