parallaxinc/Flight-Controller #44: Compass calibration needs?

MatzElectronics, 2016-09-19:
@JasonDorie For compass calibration, will it need declination and inclination input somehow, or will it "discover" these through a calibration routine?  Just curious if it has GS implications...

JasonDorie, 2016-09-19:
I'm not sure, honestly.  I suspect it'll need to be entered, but it might
be discoverable on startup / init / calibrate too.

J

RoboTechie, 2016-09-20:
As far as I know from Search and Rescue (and a little bit of googling) you can't determine declination without location (GPS) - I don't think we can do any calibration trickery to extrapolate that information. NOAA has a great [tool ](http://www.ngdc.noaa.gov/geomag-web/)for calculating declination using the WMM of IGRF models. Lots of good info available from the [NOAA Geomagnatism](https://www.ngdc.noaa.gov/geomag/) page. 

Here are a few potentially useful open-source magnetic declination code resources.
- [Magnetic Declination in C# - Blue Toque Software](http://www.bluetoque.ca/2013/01/magnetic-declination-in-c-sharp/)
- [Geomagnatism  - Natural Atlas](https://github.com/naturalatlas/geomagnetism)

JasonDorie, 2016-10-12:
Having played with this a bunch now, declination can be ignored until we have on-board GPS.  Not having a "true" north is irrelevant as long as whatever north you have remains fixed - It's just an external reference to lock onto so the heading doesn't drift.

For inclination, the current code cancels it out by projecting the inclined vector onto the earth horizontal plane, then computes the compass heading from that flattened vector.  Take a look at the "Heading Calculation" diagram mid-page here: http://ozzmaker.com/compass2/

In that image, 'H' is the inclined heading vector, pointing forward-right and downward, into the earth.  Projecting that vector to the horizontal plane gives you Hh, and that is the vector that you compute heading from (or in our case, align the IMU matrix with).

JasonDorie, 2016-11-09:
This is coded.  Needs testing, but seems to be correct, and the calibration code is handled by the quad itself.  In "FloatStreamCompiler" branch, also requires flashing with the "elev8-drivers" firmware first, hitting "enter" to upload, then flashing with main flight firmware.

JasonDorie, 2016-11-13:
I have tested this - needs verification by others.

MatzElectronics, 2016-11-15:
Tested and verified.  Before I close this - do we want to consider whether allowing the user to input an offset (declination) is necessary, especially for GPS?

JasonDorie, 2016-11-15:
It'll probably be necessary when GPS comes online, but it certainly isn't
necessary without it, and I'll also need to write the code to use it.

MatzElectronics, 2016-11-15:
Sounds good, I'll close this one and let that propagate back up when we get closer.

closed 2016-11-15 by MatzElectronics