parallaxinc/Flight-Controller #49: Investigate changing startup beeping

MatzElectronics, 2016-10-10:
Is the startup beeping when the gyro isn't settled annoying?  Should it be disabled under any or some circumstances (usb only)... can the volume be reduced?

BTW - this is a low-priority issue, more for documenting the idea than taking immediate action.

JasonDorie, 2016-10-10:
Oh dear god, seriously?

I find an erratic flight flight controller that crashes annoying, so I put
the beeps in to prevent that - if you move the controller while I'm finding
the gyro zero you get what was captured on video when Kyle / Ken tried the
throw launch, powering the quad while held.  That was when I put in the
movement detect beeps.

All the startup happens before I can tell it's connected to the PC, so I
can't do USB specific startup behavior without a lot of rework, and
probably adding a second to the boot time.

J

MatzElectronics, 2016-10-10:
Ok - I vote safety is paramount to being not annoyed :)

RoboTechie, 2016-10-11:
False Alarm. Issue Closed!
It turns out the beeping was coming from the power distribution board (not Parallax's) I was using (it was mounted directly below the FC, and it had a buzzer on it that apparently beeps every time it gets power).

closed 2016-10-11 by RoboTechie