parallaxinc/Flight-Controller #51: Validate gyro calibration

JasonDorie, 2016-11-08:
Add code to characterize the quality of gyro calibration.

- Range of temperature measurement should be broad enough for good line fit
- Range of gyro readings within measurement range should be within some max dist of chosen line
- detect bumps & auto reset?
- Display small indicator like "pending" (red) and change to "complete" (green)

JasonDorie, 2016-11-09:
Partly done - now in FloatStreamCompiler branch.  Doesn't auto reset, but quantifies noise and range.

MatzElectronics, 2016-11-15:
Tested and verified.  The only concern I have is that when I clicked "upload calibration" both bars went red.  I understand why, but docs need to explain what's happening there.

closed 2016-11-15 by MatzElectronics

JasonDorie, 2016-11-15:
I could make it put a message in there in green saying "upload successful"
or something for 5 seconds after you upload.  Less confusing.

MatzElectronics, 2016-11-15:
Sounds good, if you can, make it so!

*Matthew Matz  |  STEM/Robotics Educator*
Parallax Inc.  |  Direct: 916-625-3019  |  www.parallax.com  |  @M
atzElectronics <http://twitter.com/MatzElectronics>